xtr4nge/FruityWifi #169: Cannot install any uninstalled external modules

gh0stshell, 2015-10-31:
it shows downloading then nothing

gh0stshell, 2015-10-31:
2015/10/30 01:12:24 [error] 594#0: *1 FastCGI sent in stderr: "PHP message: PHP Fatal error: Uncaught exception 'Requests_Exception' with message 'cURL error 7: Failed to connect to 127.0.0.1 port 8081: Connection refused' in /usr/share/fruitywifi/www/modules/fruityproxy/includes/Requests/Requests/Transport/cURL.php:291

gh0stshell, 2015-10-31:
I can use curl from the CLI and connect to 8081

gh0stshell, 2015-10-31:
I am using a R Pi running Kali arm and installed FruityWifi via apt-get

gh0stshell, 2015-10-31:
I am running version 2.2 and trying any of the modules that have install listed next to them such as tor or stalker

xtr4nge, 2015-10-31:
Hi gh0stshell,
It looks like you have a problem with the core installation/setup.
This is a topic for the forum. Please open a new thread in:
http://fruitywifi.boards.net/

regards,

closed 2015-10-31 by xtr4nge