xtr4nge/FruityWifi #170: Install

gh0stshell, 2015-10-31:
If using a Rasp Pi running Kali Arm latest, can I use the repo install method or am I suppose to be using the install script method? I wonder if this is why I cannot install any non installed modules.

gh0stshell, 2015-11-01:
I am getting no responses on that board you send me to.

gh0stshell, 2015-11-01:
i can install using the apt-get option, but not the tar.gz option

gh0stshell, 2015-11-01:
if install using the scripts I can install all the modules, if i use the kali repo I cannot install modules that are not in the repo. I also cannot issue fruitywifi-module-name per your wiki.

gh0stshell, 2015-11-01:
using the scripts I also get no proxy errors in my logs, where can I see what is happening when I try installing a module using the tar.gz method? I think the log data is not showing the errors when trying to install using the tar.gz option.

gh0stshell, 2015-11-01:
able to install plugins under fruityproxy>plugins without any issues

gh0stshell, 2015-11-01:
I cannot use "Try to run the installation script manually. Each module|plugin contains an installation script. Go to the folder /usr/share/fruitywifi/www/modules/{module-name}/includes/ and execute manually the script install.sh." since I installed from the kali repo

xtr4nge, 2015-11-01:
Please go to the forum, this is not a topic for GitHub.

regards,

closed 2015-11-01 by xtr4nge