xtr4nge/FruityWifi #171: Can't connect to wifi with any device

exqua, 2015-11-12:
Hello, 
First of all. really nice project man!
But.. i have a small issue.
I can't connect wifi with any device. Wifi is up and running, SSID is visible.

Running on my Rasp B+ with v2.2 panel.

What can i do to get this working?

Greetings,
Exqua

exqua, 2015-11-13:
I see the problem, i'm having a awus036h and that one is not supported with hostapd.

xtr4nge, 2015-11-13:
Hi exqua,
I hope you are well.
Sorry for the delay in my response.

That is correct, Realtek chipsets are not supported by Hostapd.
Please take a look the following thread in the forum:
http://fruitywifi.boards.net/thread/59/suported-wlan-adapter

regards,

closed 2015-11-13 by xtr4nge

ClemaX, 2015-12-26:
I had a similar problem, it turned out being the dnsmasq config laking some informations.
I need to ad something like this:
# subnet mask
dhcp-option=1,255.255.255.0
# gateway
dhcp-option=3,192.168.1.1
# dns
dhcp-option=6,192.168.1.1

[**Source**] (http://raspberrypi.stackexchange.com/questions/33946/dnsmasq-dhcp-not-giving-ip-address)