xtr4nge/FruityWifi #172: wireless don't assing ip to client

Ramses2056, 2016-01-17:
Hi there, i have problem conecting my clients to fruitywifi. i have began with kali 2.0.1 with tft support on  RPi B+. and i downloaded master.zip installed fruitywifi and its modules with the provided scripts.
im using a RT5370 Wireless Adapter on my pi but when i try to connect my client i dont get an ip.
im using my macbook pro that its connected via wifi(with gateway 10.0.0.1) to share internet with my pi,via ethernet(thunderbolt). my mac gives my pi an ip of 192.168.2.10. 
can you please help me to start testing this cool program?
i hope that i had expain my issue, and sorry for my english

powerregist, 2016-01-19:
In Kali 2.0.*   you need to stop network-manager (service network-manager stop) and restart AP mod in Fruity

Ramses2056, 2016-01-19:
Thanks for your reply powergist.
What's the best distro for "ONLY" running fruitywifi and metasploit-framework at the same time?
any recommendations?
Im trying to avoid issues with kali2.0. everytime i get stuck in some new issue with kali linux.

closed 2016-01-19 by Ramses2056

powerregist, 2016-01-20:
Try to return to Kali 1.9. There you have such problems will not be. Or learn Kali 2.0, it is a great tool.

Ramses2056, 2016-01-20:
Ok, many thanks for your reply