xtr4nge/FruityWifi #173: installation failed on Kali 2.0

xros, 2016-01-19:
`apt-get upgrade && 
apt-get install fruitywifi -y`
kali 2.0 x86_32 system, do you know how to fix this? 
This looks like a dependency error.

with error 

<pre><code>
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 fruitywifi : Depends: fruitywifi-core (>= 2.2-0kali1) but it is not going to be installed
              Depends: fruitywifi-module-mdk3 but it is not going to be installed
              Depends: fruitywifi-module-nessus but it is not going to be installed
              Depends: fruitywifi-module-vfeed but it is not going to be installed
              Depends: fruitywifi-module-nmap but it is not going to be installed
              Depends: fruitywifi-module-whatsapp but it is not going to be installed
              Depends: fruitywifi-module-rpitwit but it is not going to be installed
              Depends: fruitywifi-module-kismet but it is not going to be installed
              Depends: fruitywifi-module-karma (>= 1.5) but it is not going to be installed
              Depends: fruitywifi-module-autostart but it is not going to be installed
              Depends: fruitywifi-module-captive but it is not going to be installed
              Depends: fruitywifi-module-responder but it is not going to be installed
              Depends: fruitywifi-module-ngrep but it is not going to be installed
              Depends: fruitywifi-module-sslstrip but it is not going to be installed
              Depends: fruitywifi-module-dnsspoof but it is not going to be installed
              Depends: fruitywifi-module-urlsnarf but it is not going to be installed
              Depends: fruitywifi-module-autossh but it is not going to be installed
              Depends: fruitywifi-module-ettercap but it is not going to be installed
              Depends: fruitywifi-module-tcpdump but it is not going to be installed
              Depends: fruitywifi-module-meterpreter but it is not going to be installed
              Depends: fruitywifi-module-nmcli but it is not going to be installed
              Depends: fruitywifi-module-supplicant but it is not going to be installed
              Depends: fruitywifi-module-3g-4g but it is not going to be installed
              Depends: fruitywifi-module-squid3 but it is not going to be installed
              Depends: fruitywifi-module-mana (>= 1.1-0kali3) but it is not going to be installed
              Depends: fruitywifi-module-phishing but it is not going to be installed
              Depends: fruitywifi-module-recon but it is not going to be installed
              Depends: fruitywifi-module-bdfproxy but it is not going to be installed
              Depends: fruitywifi-module-nginx but it is not going to be installed
              Depends: fruitywifi-module-ap but it is not going to be installed
              Depends: fruitywifi-module-fruityproxy but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

</code></pre>

xtr4nge, 2016-01-19:
Hi xros,
I didn't see that problem before.
Try installing the core only to debug the dependencies: `apt-get install fruitywifi-core`
You can also use FruityWiFi installer to get the latest version.

Please ask in FruityWiFi forum: http://fruitywifi.boards.net/

regards,

closed 2016-01-19 by xtr4nge

xros, 2016-01-23:
@xtr4nge thanks for you information.
But it still does not work in my virtualbox Kali 2.0 instance. if I tried `apt-get install fruity-core`. 
it repored

     fruitywifi : Depends: fruitywifi-core (>= 2.2-0kali1) but it is not going to be installed

same error as `apt-get install fruity-wifi`. And I asked on the forum for help. I did `apt-get dist-upgrade` but it still failed. No some hints?

xtr4nge, 2016-01-23:
I never seen that error before. I need to take a look to the updates on Kali Linux (I don't control the repository).

You can use FruityWiFi installer from GitHub:
https://github.com/xtr4nge/FruityWifi

regards,

xtr4nge, 2016-01-25:
Hi xros,
I installed the latest Kali Linux today and then FruityWiFi from the repository and I didn't get any dependencies error.

regards,

xros, 2016-01-31:
@xtr4nge hi, the actual error is a dependency error.
It said fruitywifi on kali2,

    The following packages have unmet dependencies:
    php5-curl : Depends: php5-common (= 5.6.7+dfsg-1) but 5.6.9+dfsg-0+deb8u1 is to be installed
    E: Unable to correct problems, you have held broken packages.

The package `php5-common` in kali 2 is bit newer than what fruitywifi needs, so it reported with error.

This is from a new installation of Kali 2 in virtualbox.

But if i uninstall the newer package `php5-common`, kali will automatically remove some other packages, which can destroy system.

Do you know how to coordinate?

Thanks!

xtr4nge, 2016-01-31:
Hi xros,
There is not a specific version of php5-common and/or php5-curl required by FruityWiFi.
Try installing php5-common, php5-curl, etc (check FruityWiFi installation script for reference) and then install FruityWiFi from Kali Linux repo.

You can also check the dependencies before to install: 
`apt-cache depends fruitywifi`
`apt-cache depends fruitywifi-core`
`etc...`

As I mentioned before, I did a clean install of Kali Linux after your first post and I didn't have any problem during the installation.

Please continue on the forum as more users can help you with the issue.
http://fruitywifi.boards.net

regards,