xtr4nge/FruityWifi #175: module karma not running

phobusenko, 2016-01-21:
Hi,

I was trying to run fruitywifi on my raspberry pi2, everything is OK but some modules as karma or mana are not able to run. At first I tried it on Kali 2.0 via apt repository, then I used last version from github, still the same problem.... at the end I reinstalled my raspberry to raspbian and tried version also from github, still not working. So I was digging more and found out that when I try it to start manually a get this error message:
Failed to connect to hostapd - wpa_ctrl_open: No such file or directory

root@raspberrypi:/usr/share/fruitywifi/www/modules/karma/includes# ./hostapd_cli -p /var/run/hostapd-phy0 karma_enable
Failed to connect to hostapd - wpa_ctrl_open: No such file or directory

It's problem with hostapd.

Is there any solution?

Thank you for an amazing project. :)

xtr4nge, 2016-01-21:
Hi phobusenko,
The chipset needs to be supported by Hostapd.
Please the forum for more info: http://fruitywifi.boards.net/thread/59/suported-wlan-adapter

If your chipset it is not supported by Hostapd you can use [AP] mode Airmon-ng 
(but airmon-ng option that not support Karma and/or Mana)

regards,

closed 2016-01-21 by xtr4nge

phobusenko, 2016-01-21:
Hi,

AP mode using airmon-ng is so slow that he could start to backing up any moment, known problem actually. :) Thing is, that hostpad is manually working on my both interfaces (when I was doing hotspot for myself some time ago), interfaces are:
wlan0		Ralink RT2870/3070	rt2800usb - [phy1]
wlan1		Atheros AR9271	ath9k - [phy0]

Atheros should be working properly at least. Oh, and on Kali hostapd was running just fine installed from apt fruitywifi-core. But karma/mana not...Hmmm...

Anyway I was searching two days now and I am not able to solve problem :(

phobusenko, 2016-01-21:
Soo I added kali repo to raspbian, installed everything again and AP with hostapd is working.. lol :)

xtr4nge, 2016-01-21:
Cool,
For future question please got to the forum, as you will find more people with similar problems and solutions ;)

regards,

phobusenko, 2016-01-21:
Actually sorry for that. Yes I will! Thanks.

xtr4nge, 2016-01-22:
No problem ;)

regards,