xtr4nge/FruityWifi #178: extract MAC address error

darkr4y, 2016-02-18:
My OS enviroment:
$ uname -a
Linux DarKali 4.3.0-kali1-686-pae #1 SMP Debian 4.3.5-1kali1 (2016-02-11) i686 GNU/Linux

No module installed , only [Wireless], I cannot start AP, so I reviewed the code,found that:
in file:
https://github.com/xtr4nge/FruityWifi/blob/master/FruityWifi/www/scripts/status_wireless.php
line 173

    $exec = "$bin_ifconfig -a $io_in_iface |grep HWaddr";

in my OS env ,the ifconfig command result is:

```
$ ifconfig -a wlan0     
wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 3c:46:d8:59:e8:35  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

 $ ifconfig -a wlan0:f:
wlan0:f:: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 3c:46:d8:59:e8:35  txqueuelen 1000  (Ethernet)
```

There is no "HWaddr" in result...
I think the best way to extract the MAC address is using RegEx.
:)

xtr4nge, 2016-02-20:
Hi darkr4y,
Kali Linux Rolling has a couple of changes that they are breaking FruityWiFi. I will fix the compatibility issues on v2.5.

Thanks for the heads up ;)
regards,

darkr4y, 2016-02-21:
also the command of get interfaces also has an error
in file www/page_config_adv.php line 194
```
$ifaces = exec("/sbin/ifconfig -a | cut -c 1-8 | sort | uniq -u |grep -v lo|sed ':a;N;$!ba;s/\\n/|/g'");
$ifaces = str_replace(" ","",$ifaces);
$ifaces = explode("|", $ifaces);
``` 
replace it with
```
exec("ifconfig -a | expand | cut -c1-8 | sort | uniq -u | grep -v lo| awk -F: '{print $1;}'",$ifaces);
```
and in your code, there are a lot of duplicate code
"GetIface" could be written to a function

Thx :)

xtr4nge, 2016-02-21:
Hi darkr4y,
Yes, the issue is because the output from `ifconfig` on Kali Rolling. 
I already have the fixes done and I will release a new version as soon as possible. 

regards.

xtr4nge, 2016-02-21:
Hi darkr4y,
FruityWiFi v2.4 has been released. Compatibility issues with Kali Linux Rolling has been fixed.
You also need to upgrade AP module to v1.2.

regards,

closed 2016-02-21 by xtr4nge