xtr4nge/FruityWifi #179: Driver initialization issue

TheFork1337, 2016-02-22:
So i have a raspberry pi b and a TL-WN722N usb adapter and i can get it to work with the AP and Karma but i can't start mana. I tried this guide: http://fruitywifi.boards.net/thread/65/rpi-kali-01-fruitywifi-issues 

I get this error : 
root@fruitywifi:/usr/share/fruitywifi/conf# hostapd hostapd.conf
Configuration file: hostapd.conf
nl80211: Could not configure driver mode
nl80211 driver initialization failed.
hostapd_free_hapd_data: Interface wlan0 wasn't started

darkr4y, 2016-02-23:
the same problem i have...

xtr4nge, 2016-02-23:
Hi guys, 
I guess that you are on Kali Linux, you can try 2 alternatives:

Install the version from the repository:

`apt-get install fruitywifi-module-mana`

or download the version from the GitHub repository ( from the modules page ), and change the following in the installer. (`/usr/share/fruitywifi/www/modules/mana/includes/install.sh`)

`lsb_release -c |grep "jessie"` to `lsb_release -c |grep -iEe "jessie|kali"`

regards,

TheFork1337, 2016-02-23:
Thanks for the quick respons. No Im currently running a clean install of rasbian. 

regards

xtr4nge, 2016-02-23:
Hi TheFork1337,
Raspbian should not give you any trouble with that if the wifi dongle is supported and the drivers are installed.

Are you using FruityWiFi v2.4?
Are you having the problem only with Mana?

regards,

TheFork1337, 2016-02-23:
Im using Fruitywifi v2.2 that was the only one i got when i downloaded /archives/master.zip

Does mana require something else exept AP to be activated?

xtr4nge, 2016-02-23:
The latest FruityWiFi is v2.4 and you can find it in the GitHub repository.
It contains a few new and fun features in the AP module, so upgrade to v2.4 if you have a chance ;)

About Mana, to activate mana you need to have the module mana installed and the [AP] mode set as hostapd-mana and the same for Karma (you need to setup this in the config page). Then start the AP from the status page, and after the AP is running, you can enable mana ( or Karma if you are using hostapd-karma )

regards,

darkr4y, 2016-03-01:
here I hv a problem.
Env:  Ras Pi 2 / Raspbian Latest / FruityWifi 2.4
When I click the [AP] button in status page,
after refresh,it shows no running.. but in config page it shows running..
and the true is no Wifi Ap is set up...

closed 2016-09-05 by xtr4nge