xtr4nge/FruityWifi #181: Can not enable AP or Karma (Hostapd)

skordonis, 2016-03-04:
Hello! I installed Fruity Wifi on a freshly installed raspbian and i i tried to start the AP from the fruitywifi admin panel but after it loads nothing changes. The same thing happens when i try to enable karma. I set up the config section (IN-OUT/ELAN-ETH).
My USB Wifi dongle is a Edison with chipset Ralink 5370! Any ideas?

xtr4nge, 2016-03-04:
Hi skordonis,
There are a few basic checks that you need to do, as Ralink RT5370 is supported by hostapd on RPi.

Without starting AP mode [`hostapd-karma`] you can't start Karma
Without starting AP mode [`hostapd-mana`] you can't start Mana.

Check the FAQ
http://fruitywifi.boards.net/thread/52/faq

Install firmware-ralink
`apt-get install firmware-ralink`

Assuming that you are using wlan0 and eth0, go to config [menu]:
Set `IN` as `wlan0`
Set `OUT` as `eth0`

Remove `wlan0` entries from `/etc/network/interfaces`

Alternatively: 
unlock interfaces: `rfkill unblock wifi; sudo rfkill unblock all`
stop network manager `/etc/init.d/network-manager stop`

you can also search and ask in the forum for more info: http://fruitywifi.boards.net

let me know.
regards

grayburn, 2016-03-05:
I am seeing similar. Using the latest version of Raspbian (2016-2-26) with Fruity 2.4 and if I select hostapd, the AP module doesn't start. But if I use Hostapd-karma, then the AP module works fine. Using Airmon-ng works as well.

skordonis, 2016-03-07:
Done! Thanks!

xtr4nge, 2016-03-12:
No prob ;)

closed 2016-03-12 by xtr4nge