xtr4nge/FruityWifi #186: AP/Karma not working??

FinlayDaG33k, 2016-04-25:
When I try to setup an AP using FruityWiFi, 
it says that [AP] is enabled, but when I go to my laptop and check it, the AP is not listed.
I have no clue how the options should be as the documentation is quite outdated and bare (webUI say IN-OUT instead of Internet-Wifi for example)

my config at the moment is this:

- MODE: IN - --- [AP]
- [AP]: Hostapd-Karma
- IN:
  - Interface: wlan2
  - [Manual]
  - IP: 192.168.101.1
  - Netmask: 255.255.255.0
  - Gateway: empty
- stop [on]

xtr4nge, 2016-04-27:
Hi FinlayDaG33k,
Your wireless card needs to be supported by Hostapd. Please go to the forum for more details and 

Forum: http://fruitywifi.boards.net
FAQ: http://fruitywifi.boards.net/thread/52/faq
Supported adapters (Hostapd): http://fruitywifi.boards.net/thread/59/suported-wlan-adapter

regards,

closed 2016-04-27 by xtr4nge

FinlayDaG33k, 2016-04-27:
I found the issue that caused it: 
I had selected my onboard WiFi (an unsupported adapter) instead of my USB adapter (which is supported)

FinlayDaG33k, 2016-04-27:
Unfortunately it didn't get solved.

I now have setup my `TL-WN722N` with FruityWifi, but the AP still doesn't show up anywhere (my tablet, my PC, my cellphone).

I'm using Kali2

xtr4nge, 2016-04-27:
Check all the links and recommendations on the mentioned post.
I want to keep GitHub just for bugs, and this is a setup issue.

Search on the forum or post a new thread if required (on the forum).

regards,

FinlayDaG33k, 2016-04-27:
After doing some tinkering, I managed to get the AP to work.
for somekind of unknown reason, the AP said it was on, but infact, it wasn't.

xtr4nge, 2016-04-27:
Good, as I mentioned on the other issue, there are some services that can cause troubles, dnsmasq, network-manager, setup on interfaces, etc.

For further questions, please go to the forum ;)

regards,