xtr4nge/FruityWifi #188: Two Problems about Single-NIC Wifi Repeating

Cnly, 2016-05-01:
Hi, I'm trying to create an AP on the same Wifi dongle(creating two virtual interfaces) connected to another wireless network. Now I have succeeded, but during the process I found some problems with the AP module.

0. Many dongles require **the same channel** be set for both the station and the AP. But for now the channel used for AP is hard-coded in `hostapd.conf` and `hostapd-secure.conf`. Hope there can be something doing the configuration automatically, or something making this easier...

0. The AP module is making the entire dongle `phy0` unmanaged in `NetworkManager.conf`, rather than the virtual interface `phy0_ap` to create AP on, which will disconnect `phy0_sta` from the current network. I found that the function `setNetworkManager` is obtaining MAC addresses through  `macchanger --show $io_in_iface |grep 'Permanent'`. Virtual interfaces have different MAC addresses from `phy0`'s. Changing `Permanent` to `Current` fixed this issue.

xtr4nge, 2016-05-04:
Hi Cnly, I hope you are well.

At the moment, the channel can only be changed from the config file, but I'm planning to add that and other options on the next release.
Stopping Network-Manager should solve any issue related with Network-Manager, but I will take a look.

Thanks for your comments ;)
xtr4nge

Cnly, 2016-05-07:
Thanks for your reply! Maybe you can take https://github.com/oblique/create_ap as a reference.