xtr4nge/FruityWifi #190: php_modules.php?show not showing any modules

Bry-fi, 2016-05-20:
I have installed fruitywifi 2.4 with no modules and i try to list the modules and nothing loads on the page.

simonpunk, 2016-06-03:
Same situation here, I have followed the latest instruction to install the fruitywifi 2.4 and its all modules, but at the panel it still showing me no modules are installed. Don't know what happened.

xtr4nge, 2016-06-03:
Hi guys, this is not a bug.
Please go to the forum and open a new thread (I need more info). I will help you there.
(What device, What os version, Debian, Kali, How did you installed FruityWiFi, etc...)

simonpunk, you need to be sure that the setup for the user fruitywifi is correct, please check this thread:
http://fruitywifi.boards.net/thread/111/fruitywifi-v2-4

Note: If you are on Kali Rolling, you just need to use **apt-get install**

regards,

closed 2016-06-03 by xtr4nge