xtr4nge/FruityWifi #194: Internet+AP mode in one wireless card

ZapperC, 2016-06-08:
Hi,
I want to use FruityWifi in Evil Twin attack and i have a plan about it but i want to know if it will work or not.
I have two wireless cards and both are TL-WN722N . What i want to do is to use the first card ( wlan0 ) as the internet source and FruityWifi AP in the same time and i will put the second card ( wlan1 ) in monitor mode for DoSing and jamming the target AP to make the victim connect to my FruityWifi AP ( which was created on wlan0 ).
Will this plan work or i'm missing something here ?

xtr4nge, 2016-06-08:
Hi ZapperC,
Sure, that is standard setup for FruityWiFi, and also you can setup and triggered everything from FruityWiFi web-interface.
Go to Menu/Config and setup [IN[ for the AP interface, [OUT] is your internet connection interface. 
For the 3rd interface you need to setup [Monitor Interface], select your 3rd iface from the pulldown and press start, you will see mon0 in green. (you can perform this step before or after starting the AP module)
Then you are ready to start FruityWiFi AP (AP module). 
You can use the module MDK3 to deauth (jamming) devices. (You need the 3rd iface on monitor mode as mon0, that is why you need to do the previous steps)

regards,

neekolaz, 2016-08-28:
This may help a bit.  I have been all over with trying to fix the AP issues with my Ralink RT5370.  I had to completely remove network-manager with 2016-05-26-raspian-jessie. using the following command
`$  /etc/init.d/network-manager stop
$ update-rc.d network-manager remove`
i restarted networking, and was able to fully use the AP mode.  I hope this helps, and xtr4nge this is an amazing project, I hope to be proficient enough to actually contribute soon.

xtr4nge, 2016-09-18:
Network-Manager can be disabled when the AP starts. Check the options on AP module, tab DHCP-DNS.

regards,

closed 2016-09-18 by xtr4nge