xtr4nge/FruityWifi #196: FruityWifi AP connection is not working

riramar, 2016-07-07:
I'd like to run FruityWifi in my CHIP (https://getchip.com) so I follow the "ARM version" installation steps.
Everything seems to be installed properly. After started the Wireless service I tried to connect to FruityWifi wireless network from my mobile but it never connects. I can see only "Connecting" from my mobile. Below the syslog relevant part.
It seems the FruityWifi DHCP is not providing any IP to the clients.
Any tips how to fix that?

```
Jul  7 16:44:17 chip hostapd: wlan1: STA ac:5f:3e:75:42:44 IEEE 802.11: associated
Jul  7 16:44:17 chip hostapd: wlan1: STA ac:5f:3e:75:42:44 RADIUS: starting accounting session 577E85F2-00000002
Jul  7 16:44:48 chip hostapd: wlan1: STA ac:5f:3e:75:42:44 IEEE 802.11: disassociated
Jul  7 16:44:48 chip rsyslogd-2007: action 'action 17' suspended, next retry is Thu Jul  7 16:45:48 2016 [try http://www.rsyslog.com/e/2007 ]
Jul  7 16:44:49 chip hostapd: wlan1: STA ac:5f:3e:75:42:44 IEEE 802.11: associated
Jul  7 16:44:49 chip hostapd: wlan1: STA ac:5f:3e:75:42:44 RADIUS: starting accounting session 577E85F2-00000003
```

xtr4nge, 2016-07-11:
Hi riramar,
Unfortunately I don't have any experience with CHIP, but I saw I forum were people is playing with CHIP + FruityWiFi 
Please take a look here: https://bbs.nextthing.co/t/fruity-wifi-running-on-pocketchip/5513

Other than that, you can check on the [AP] module if all the required services are running [AP, DHCP, DNS]. Verify that you don't have any filter, and check the DNSmasq (FruityWiFi) log. (/usr/share/fruitywifi/logs/dnsmasq.log)

regards,

closed 2016-07-11 by xtr4nge

riramar, 2016-07-21:
Thanks a lot for the link! I'll try to install following those instructions and let you know if everything works fine. Maybe we can include a section in your wiki for CHIP. :)