xtr4nge/FruityWifi #197: install script does not add fruity to init.d [2.4]

Mesaman, 2016-07-12:
When installing the newest version by install script fruity is not added to the init.d processes. You have to start nginx and php manually to access fruitywifi

xtr4nge, 2016-07-12:
Hi Mesaman,
What OS/Hardware are you using?
The installation scripts execs the following lines:

```
update-rc.d nginx defaults
update-rc.d php5-fpm defaults

/etc/init.d/nginx restart
/etc/init.d/php5-fpm restart

```

Mesaman, 2016-07-12:
Hi,

im on raspbian 8 and a colleague did it on kali. Both of us had to start nginx and php manually after rebooting the system. We were trying to start fruity with /etc/init.d/fruitywifi start but there is no such process when installing the 2.4 version. Maybe the wiki should be updated for version 2.4?

xtr4nge, 2016-07-12:
You only get **/etc/init.d/fruitywifi** when is installed using apt-get (Kali Linux only). It is a init script created by the Kali Linux team for FruityWiFi on Kali (but you also need to start php5-fpm).

The installation using GitHub version, needs to start **nginx** and **php5-fpm**.
I'm asking about your OS/hardware as it looks like **update-rc.d** is not having any effect on latest raspbian (RPi), but works fine on Debian base.

closed 2016-08-21 by xtr4nge