xtr4nge/FruityWifi #205: DHCP-DNS page not saving settings

FinlayDaG33k, 2016-10-03:
When I go to my DHCP-DNS page (`/modules/ap/`) the settings I fill in, don't save at all

xtr4nge, 2016-10-03:
Hi FinlayDaG33k,
This could be a permissions problem. FruityWiFi execs several commands using sudo (user fruitywifi).
You need to: 
- check that Nginx is running as fruitywifi user 
- check sudo setup for fruitywifi user.
- check folder and files permissions (/usr/share/fruitywifi/modules/ap/)

What hardware and OS are you using?
How did you install FruityWiFi? 

The installation script is doing all the permissions setup for you, and of course the same if you are using Kali Linux repo (apt-get...)

regards,

FinlayDaG33k, 2016-10-03:
I think it's running as fruitywifi (`top -U fruitywifi` shows a few nginx processes)

I'm running on `Raspbian Jessie`.
when running `ls -l /usr/share/fruityiwifi/modules/ap/` I get `No such file or directory`.
However, when I do `ls -l /usr/share/fruityiwifi/www/modules/ap/` I see it's owned by `root`.

Upon trying more stuff from the FruityWifi suite, I noticed that the DNSspoof also doesn't work.

I've tried FruityWifi on my laptop (with ParrotOS) with great success, so I think it's the combination of Pi and Fruitywifi that goes wrong...

FinlayDaG33k, 2016-10-03:
Oh yea, I installed FruityWifi with the [installer script](https://github.com/xtr4nge/FruityWifi/blob/master/install-FruityWiFi-PHP7.sh)

FinlayDaG33k, 2016-10-03:
After doing a little test, I confirmed that it's most likely a configuration issue.
when running the AP module with only IN, no OUT, I can still ping the internet trough the AP.

xtr4nge, 2016-10-03:
If you are using Raspbian, you need to use **install-FruityWiFi.sh** and not install-FruityWiFi-PHP7.sh
I created install-FruityWiFi-PHP7.sh version for Kali Linux as they remove php5 from the repo.

Note: yes the correct path is: /usr/share/fruitywifi/www/modules/ap/
regards,

FinlayDaG33k, 2016-10-04:
ah okay :) I will reinstall Fruitywifi with the other one instead :) will report back to you

xtr4nge, 2016-10-04:
Cool, yes, let me know ;)
Thanks

FinlayDaG33k, 2016-10-04:
Reinstalled it with `sudo ./install-FruityWiFi.sh` , still the same issue :\ 
Here is a screenshot of my config:
[image]

And again, when I try to ping a server, with this config, FruityWifi still let's traffic trough to the interwebs.

FinlayDaG33k, 2016-10-04:
Changing the config to the following made me able to "block" all traffic to the internet:
[image]

Still trying to figure out why the DNSSpoof plugin doesn't spoof anything at all...

xtr4nge, 2016-10-04:
Hi FinlayDaG33k,
But I see that you can save the settings. It was not that your problem?

About DNSspoof, that module has been deprecated. You need to use FruityDNS module for DNS spoofing.

The functionality for MODE has not been implemented yet. The change on the MODE (IN-OUT, IN-NONE, etc) doesn't have any impact (at the moment) in the setup. For now, the setup always uses IN Interface and OUT interface doesn't matter the MODE that you are assigning.

Please for more questions about the setup go to the forum ;)
http://fruitywifi.boards.net

regards,

closed 2016-10-04 by xtr4nge