xtr4nge/FruityWifi #211: 3rd interface mon0 ?

hadiabounader, 2016-11-22:
Dear, 

I'm having issues setting up the interfaces.. 
if IN is for AP interface (wlan1 in my case) and OUT for the internet interface.. ( wlan0) in my case)

what's the interface for mon0 ? 
everytime i set it up to wlan1 or wlan0 it disconnects me from the network , help!

xtr4nge, 2016-11-22:
Hi hadiabounader,
You need to setup and start **Monitor Interface** in MENU/Config, is below IN and OUT config

regards,

closed 2016-11-22 by xtr4nge

hadiabounader, 2016-11-22:
I Mean Mon0 should be equal to the Access point (Wlan1) ?

xtr4nge, 2016-11-24:
No, mon0 need to be a 3rd interface, it is not related with the interface **IN** being used for the Access Point.
The interface (Monitor Interface section) mon0 is used by some modules such as WiFiRecon, DetectDeauth, DetectRogue, MDK3, Kismet, and other modules that requires an interface in monitor mode.

regards,

hadiabounader, 2016-11-24:
in my setup i have my raspberry built in wifi connected to the router ( internet), Ralink 5370 ( AP ) 
where am i supposed to put mon0 ?

xtr4nge, 2016-11-24:
FruityWiFi creates mon0 when you enable "Monitor Interface" (3rd interface, setup below IN|OUT)
In your case wlan0 is OUT, wlan1 is IN.
If you don't have a 3rd interface to use modules that requires Monitor Mode, then you don't need mon0

regards,

hadiabounader, 2016-11-25:
I need to use FruityDNS + Bettercap do i need mon0 ?

xtr4nge, 2016-11-27:
No, you don't need mon0 for these 2 modules.
If you are having trobles with FruityDNS and Bettercap is because of missing dependencies. 
Go to the forum http://fruitywifi.boards.net/

regards,