xtr4nge/FruityWifi #216: Mana and Karma cant compile after updating kali nethunter packages

jdoncouse, 2017-01-02:
Libssl gets updated to a new version that throws errors in compiling mana and karma. Had to install libssl1.0-dev again to be able to compile correctly again

xtr4nge, 2017-01-03:
Hi jdoncouse, Thanks for the info, I will take a look.

jdoncouse, 2017-01-03:
welcome. Also trying to figure out another weird issue i'm having with my nethunter nougat install. I posted about it on the forums. The ap page is showing disabled on everything except mana even though everything is working. I hit start on ap then it loads.. i see my card start to work correctly and then ap, dhcp, dns, show disabled *although it's working* then when i start mana it shows enabled.. Then same thing for Picker,Scatter,Polite shows all disabled even though i can get them working. I never get these issues installing on marshmallow based cyanogenmod roms. Idk what the deal is with this on nougat based ones.

Obakemono, 2018-02-21:
I had the same issue, the problem is with tls_openssl but amanged to solve it reverting to libssl1.0-dev as @jdoncouse said