xtr4nge/FruityWifi #217: Start ap issue

arshiay, 2017-01-07:
Hi
Thanks for script 
I have an issue 
I install this on kali on raspberry pi 3 and after boot and start php and fruitywifi AP shows in wifi menu in other devices but cant connect to
So i have to stop and star AP through browser 
My raspberry does have screen so i need AP to connect to it to control it
Is there any command to stop and start AP with terminal (i can put it in script to run at boot)
Or any solution?

xtr4nge, 2017-01-07:
Hi arshiay,
You can use Automaton (http://fruitywifi.boards.net/thread/133/module-automaton) module or FruityShell (http://fruitywifi.boards.net/board/8/fruityshell).
Automaton uses the FruityWiFi API to start/stop modules or to execute any other command or module.
Please go to the forum: http://fruitywifi.boards.net/

closed 2017-01-07 by xtr4nge

arshiay, 2017-01-07:
Thanks for reply 
I set new template in automation 
And set ssid 
Set protected to 1
And have a question how set my password ?

xtr4nge, 2017-01-08:
For a secure AP you need to add the following to the Automaton template:
`SET /config/core/hostapd_secure/1`
`SET /config/core/hostapd_ssid/AP_NAME_HERE`
`SET /config/core/hostapd_wpa_passphrase/PASSPHRASE_HERE`

regards,