xtr4nge/FruityWifi #218: Incompatibility between FruityProxy (FruityWifi V2.4) and mitmproxy 0.18.2

gpim999, 2017-01-10:
Hello,

It seems that there is an incompatibility between FruityProxy and recent versions of mitmproxy. With my current configuration (up to date Kali 2 with FruityProxy and mitmproxy installed from packages), the FruityProxy plugin is not able to start (from the web GUI). When I try to run it manually from the fruityproxy.py file, I get the following error message:

Traceback (most recent call last):
  File "./fruityproxy.py", line 106, in <module>
    config = proxy.ProxyConfig(port=listenPort, mode=proxyMode)
TypeError: __init__() got an unexpected keyword argument 'mode'

It seems to be an incompatibility with the way that mitmproxy expects options to be passed.

Best regards,
Pim.

xtr4nge, 2017-02-28:
Hi gpim999,
Thanks for the report. I will updated the modules as soon as possible.

regards,

xtr4nge, 2017-06-24:
Hi gpim999, 
For now keep the version on 0.17.
http://fruitywifi.boards.net/thread/193/mitmproxy-bdf-fruityproxy-debian-8

I will update FruityProxy to support the latest MITMPROXY (Python3) soon.

regards,

closed 2017-06-24 by xtr4nge