xtr4nge/FruityWifi #222: Cant Install fruitywifi from kali repository

JoeNibe, 2017-03-10:
When i try to install fruitywifi using
apt-get install fruitywifi
it gets stuck at 86% every time. Finally I figured it out that it is not able download few modules. So I installed fruitywifi-core and separately installed the modules. Please fix it....

pctripsesp, 2017-04-04:
apt-get install fruitywifi worked for me nice in Kali 2

xtr4nge, 2017-04-23:
Hi JoeNibe,
I suggest to not install all the modules at once. 
Using `apt-get install fruitywifi` will install the core system and all the modules (more than 50 modules). If you are new on FruityWiFi, I will suggest to start installing the core `apt-get install fruitywifi-core` (test that is working as expected), and then the modules that you need `apt-get install fruitywifi-module-{name}` (you can also install modules from the web-interface). 

regards,

closed 2017-04-23 by xtr4nge