xtr4nge/FruityWifi #225: Modules dont work.

syntax3043, 2017-03-20:
Hi guys.

First off all i installed raspbian with pixel cleanly on my raspberry pi 3 model B . After that i downloaded and run the script "sudo ./install-FruityWifi.sh" cleanly with no issues when i log on to interface of fruity wifi from "localhost:8000" i see no modules installed so i run "sudo ./install-modules.sh" again without any issue .After that modules was there but when ever i tried to enable [AP] it says "loading fruit" but it does not enable it.I tried each of modules only fruitydns managed to remain at enable however it will get disabled right after when I try to enable [AP]. In config ,I set up IN--OUT [AP]  hostapd. IN is internal wifi card from raspberry OUT is my external wifi card TL-WN422G TP-LINK. I couldn't figure it out what I am doing wrong.I formatted my sd card 5 times,tried to plug ethernet cable for "IN" while out is external wifi card nothing seems to work.

Supersqualo, 2017-06-12:
Hi i have same issue, how it resolve?

xtr4nge, 2017-06-24:
Hi guys, please check the forum:
http://fruitywifi.boards.net/thread/192/raspberry-pi-3-working
http://fruitywifi.boards.net/thread/185/fruitywifi-rpi-zero-ethernet-gadget

regards,

closed 2017-06-24 by xtr4nge