xtr4nge/FruityWifi #226: Suggestion for a new module

cezarlacatus, 2017-03-23:
Hello i was wondering if you guys could implement a tool called fluxion wich is a new way in cracking the most advanced wifi passwords.

I personally i have ordered an rasperry pi for this tool,and another ideea camed in mind(of course if isnt already implemented) you could provide an terminal for the user to install their own tools using apt get install and use them inside web-ui???
Anyway i love your work and fluxion is a tool wich MUST BE included.Is ultra-efficient
Thank you and sorry for my bad englesh.

stefjoe12, 2017-04-18:
agree

xtr4nge, 2017-04-23:
Hi cezarlacatus,
I will start working on this as soon as posible (probably next month).
Please check the forum as there are people already discussing about fluxion.

regards,

strasharo, 2017-04-23:
Since @deltaxflux took the repository down, you can use my mirror here - https://github.com/strasharo/fluxion

Let me know if I can help somehow with the integration.

deltaxflux, 2017-04-23:
Maybe you publish the latest version @strasharo

strasharo, 2017-04-23:
I've mirrored the latest state of the git repo that I had at my disposal. Added you as a contributor, feel free to push it there.

strasharo, 2017-04-29:
Fluxion is available here:
https://github.com/FluxionNetwork/fluxion

Let us know if we can assist somehow with the integration.