xtr4nge/FruityWifi #243: Connected AP client unable to resolve DNS

coolgeek6667, 2018-07-23:
I am successfully able to start AP and connect to it from my phone, but it reports "No Internet Connection" - after further investigation the issue appears that it simply is unable to resolve DNS; If I enter an IP, it can successfully access it.

My configuration:
Kali Linux (x86), installed using apt install
MODE:  IN - OUT | [AP]
{AP}   :   Hostapd (also tried Hostapd-Karma with same result)
IN: wlan0
     [Manual]
     10.0.0.1
      255.255.255.0
      <blank GW>

OUT: eth0
         [Current]
         192.168.1.170

I have tried switching the DNS to FruityDNS, but still no luck.
The DNS log do show the expected entries `[2018-07-23 18:34:20] 10.0.0.80: proxying the response of type 'A' for wifi-test.mobidia.com`, but I'm guessing the response never gets to the client.

Any ideas what I may have mis-configured?

H4xl0r, 2020-02-08:
set gateway to 10.0.0.1 in config
be sure fruitydns is working correctly 
got it working on my pi

rlvilla, 2020-06-21:
> set gateway to 10.0.0.1 in config
> be sure fruitydns is working correctly
> got it working on my pi

I'm trying to set up FruityWifi with Kali 2020.2 ARM 64 bits on RPi 3. I have problems installing dependencies for fruitydns after having similar issues with FruityWifi itself, but I think I fixed these. "install.sh" file tries installing "python-ipy python-flask python-configobj" from apt, but they aren't anymore on Kali's apt repo. I guess that's why I cant enable Fruitydns. I installed those packages using pip but I got no luck. Any ideas?

rlvilla, 2020-06-22:
I solved these dependencies by using "sudo pip install" instead of "pip install" as the python2 env used is root's, not my user's. Now I can get fruitydns to work manually but from FruityWifi web it gets disabled right after enabling. 
I saw somewhere that API module is required, I will try that way.