xtr4nge/FruityWifi #248: FruitWifi on Raspberry Pi cannot find SSID

momaydopod, 2018-10-04:
Hi Guy

I running FruitWifi on Raspberry Pi 3 B. I try to start Wireless service. But on notebook cannot find SSID FruitWifi.  Can you help me to solve problem ? 

[image]

momaydopod, 2018-10-04:
Information

[image]

xtr4nge, 2018-10-04:
Hi momaydopod,
Are you using the latest Raspbian? Are you using the built in wireless card?
If so, there are some changes that needs to be done for Stretch and FruityWiFi to work nicely together.
I put a message on the the README file about using systems based on Debian Jessie for now (latest Raspbian is based on Stretch).

It is not a big issue, but some setup needs to be changed on Strecth (I didn't have time yet to work on this) as there are some conflicts. (mainly network-manager and other network setup I can imagine)
Check these 2 posts for ideas if you want to work on tunning Stretch: http://fruitywifi.boards.net/board/11/tutorials

About the wireless cards, not all wifi cards are compatible with Hostapd (built in RPI should work), but if you are using an external card, check this post:
http://fruitywifi.boards.net/thread/59/suported-wlan-adapter
You can also change Hostapd to airmon-ng from the config page on FruityWiFi, but some modules will not work with airmon-ng.

regards,

momaydopod, 2018-10-04:
Are you using the latest Raspbian?  Yes, Linux raspberrypi 4.14.70-v7+ #1144 SMP Tue Sep 18 

Are you using the built in wireless card? I have on board wireless and usb external wireless(It's can use on Kali)
[image]

momaydopod, 2018-10-04:
My USB external wireless is Ralink RT5370

[image]