xtr4nge/FruityWifi #249: Karma Module not installing

elangovarun95, 2018-11-08:
Karma Module not installing and shows the error below : 

  Installing...
..DONE..
Makefile:780: recipe for target '../src/drivers/driver_nl80211.o' failed
0ae26b335e35ff8da1a40512206a3fbee8efc132
Archive: hostapd-karma.zip
installing Hostapd/Karma...

is only available from another source
This may mean that the package is missing, has been obsoleted, or
Package libnl-dev is not available, but is referred to by another package.
Reading state information...
Building dependency tree...
Reading package lists...
0 upgraded, 0 newly installed, 0 to remove and 37 not upgraded.
Use 'sudo apt autoremove' to remove them.
ruby-binding-of-caller ruby-coffee-script-source
libc-ares2 libhttp-parser2.7.1 libjs-coffeescript libjs-uglify nodejs-doc
The following packages were automatically installed and are no longer required:
hostapd is already the newest version (2:2.6-15ubuntu2.1).
Reading state information...
Building dependency tree...
Reading package lists...
Reading state information...
Building dependency tree...
Reading package lists...
0 upgraded, 0 newly installed, 0 to remove and 37 not upgraded.
Use 'sudo apt autoremove' to remove them.
ruby-binding-of-caller ruby-coffee-script-source
libc-ares2 libhttp-parser2.7.1 libjs-coffeescript libjs-uglify nodejs-doc
The following packages were automatically installed and are no longer required:
Reading state information...
Building dependency tree...
Reading package lists...
installing Hostapd/Karma Dependencies...

jmadden173, 2020-01-17:
I am having a similar issue with installing the Karma module. I have tried installing the module through the website then tried to manually install through the terminal but I get the following error when compiling the hostapd-karma-master repo.

```
../src/crypto/tls_openssl.c: In function ‘tls_init’:
../src/crypto/tls_openssl.c:745:2: warning: ‘TLSv1_method’ is deprecated [-Wdeprecated-declarations]
  ssl = SSL_CTX_new(TLSv1_method());
  ^~~
In file included from /usr/include/openssl/e_os2.h:13,
                 from /usr/include/openssl/ssl.h:15,
                 from ../src/crypto/tls_openssl.c:23:
/usr/include/openssl/ssl.h:1877:1: note: declared here
 DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_method(void)) /* TLSv1.0 */
 ^~~~~~~~~~~~~~~~~~
../src/crypto/tls_openssl.c: In function ‘tls_deinit’:
../src/crypto/tls_openssl.c:783:3: warning: ‘ERR_remove_state’ is deprecated [-Wdeprecated-declarations]
   ERR_remove_state(0);
   ^~~~~~~~~~~~~~~~
In file included from /usr/include/openssl/e_os2.h:13,
                 from /usr/include/openssl/ssl.h:15,
                 from ../src/crypto/tls_openssl.c:23:
/usr/include/openssl/err.h:261:1: note: declared here
 DEPRECATEDIN_1_0_0(void ERR_remove_state(unsigned long pid))
 ^~~~~~~~~~~~~~~~~~
../src/crypto/tls_openssl.c: In function ‘tls_load_ca_der’:
../src/crypto/tls_openssl.c:1285:40: error: dereferencing pointer to incomplete type ‘SSL_CTX’ {aka ‘struct ssl_ctx_st’}
  lookup = X509_STORE_add_lookup(ssl_ctx->cert_store,
                                        ^~
../src/crypto/tls_openssl.c: In function ‘tls_connection_get_keys’:
../src/crypto/tls_openssl.c:2293:24: error: dereferencing pointer to incomplete type ‘SSL’ {aka ‘struct ssl_st’}
  if (ssl == NULL || ssl->s3 == NULL || ssl->session == NULL)
                        ^~
../src/crypto/tls_openssl.c: In function ‘tls_connection_resumed’:
../src/crypto/tls_openssl.c:2545:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
../src/crypto/tls_openssl.c: At top level:
../src/crypto/tls_openssl.c:2976:1: fatal error: opening dependency file ../src/crypto/tls_openssl.d: Permission denied
 }
 ^
compilation terminated.
make: *** [Makefile:780: ../src/crypto/tls_openssl.o] Error 1

```

H4xl0r, 2020-02-13:
get my version !

haodongj, 2020-04-20:
> get my version !

I'd like to try your version. But where are the modules' source code stored in Fruitywifi? Where should I put your code?

Walvis-Sudo, 2020-09-20:
Hi  jmadden173,

were you able to fix it? I have exactly same issue using FruityWifi 2.4 and last 2020 Raspberrian OS..

matiarman, 2021-07-16:
> get my version !

I tried and I found the same errror:
installing Hostapd/Karma Dependencies...

Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
Nota, seleccionando «gcc-4.7-doc» para la expresión regular «gcc-4.7»
Nota, seleccionando «gcc-4.7-base» para la expresión regular «gcc-4.7»
gcc-4.7-doc ya está en su versión más reciente (4.7.4-1).
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 0 no actualizados.
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
E: No se ha podido localizar el paquete g++-4.7
E: No se pudo encontrar ningún paquete usando «*» con «g++-4.7»
E: No se pudo encontrar ningún paquete con la expresión regular «g++-4.7»
update-alternatives: error: la ruta alternativa /usr/bin/gcc-4.7 no existe
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
hostapd ya está en su versión más reciente (2:2.7+git20190128+0c1e29f-6+deb10u3).
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 0 no actualizados.
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
libnl-dev ya está en su versión más reciente (1.1-8).
libnl1 ya está en su versión más reciente (1.1-8).
libssl-dev ya está en su versión más reciente (1.1.1d-0+deb10u6+rpt1).
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 0 no actualizados.

Getting Hostapd/Karma...

Clonando en 'hostapd-karma'...
remote: Enumerating objects: 511, done.
remote: Counting objects: 100% (511/511), done.
remote: Compressing objects: 100% (439/439), done.
remote: Total 511 (delta 66), reused 511 (delta 66), pack-reused 0
Recibiendo objetos: 100% (511/511), 1.50 MiB | 1.58 MiB/s, listo.
Resolviendo deltas: 100% (66/66), listo.
--------------------------------
ADDING: CONFIG_LIBNL32=y / SLL TLS patch)
--------------------------------
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
Los siguientes paquetes se ELIMINARÁN:
  libnl-dev
Se instalarán los siguientes paquetes NUEVOS:
  libnl-3-dev libnl-genl-3-dev
0 actualizados, 2 nuevos se instalarán, 1 para eliminar y 0 no actualizados.
Se necesita descargar 0 B/111 kB de archivos.
Se utilizarán 466 kB de espacio de disco adicional después de esta operación.
(Leyendo la base de datos ... 171722 ficheros o directorios instalados actualmente.)
Desinstalando libnl-dev (1.1-8) ...
Seleccionando el paquete libnl-3-dev:armhf previamente no seleccionado.
(Leyendo la base de datos ... 171662 ficheros o directorios instalados actualmente.)
Preparando para desempaquetar .../libnl-3-dev_3.4.0-1_armhf.deb ...
Desempaquetando libnl-3-dev:armhf (3.4.0-1) ...
Seleccionando el paquete libnl-genl-3-dev:armhf previamente no seleccionado.
Preparando para desempaquetar .../libnl-genl-3-dev_3.4.0-1_armhf.deb ...
Desempaquetando libnl-genl-3-dev:armhf (3.4.0-1) ...
Configurando libnl-3-dev:armhf (3.4.0-1) ...
Configurando libnl-genl-3-dev:armhf (3.4.0-1) ...
[Patching completed]

Building Hostapd/Karma...

  CC  main.c
[... 125 lines cut ...]
  CC  ../src/crypto/sha1-internal.c
  CC  ../src/crypto/sha1-pbkdf2.c
  CC  ../src/crypto/sha1-tlsprf.c
  CC  ../src/crypto/md5-internal.c
  CC  ../src/crypto/rc4.c
  CC  ../src/crypto/random.c
  CC  ../src/ap/wmm.c
  CC  ../src/ap/ap_list.c
  CC  ../src/ap/ieee802_11.c
  CC  ../src/ap/hw_features.c
  CC  ../src/drivers/driver_common.c
  LD  hostapd
  CC  hostapd_cli.c
  CC  ../src/common/wpa_ctrl.c
  CC  ../src/utils/edit_simple.c
  LD  hostapd_cli

Copying Hostapd/Karma...

Cleaning Up...

./install.sh: línea 54: cd..: orden no encontrada
./install.sh: línea 55: cd..: orden no encontrada
rm: no se puede borrar 'hostapd-karma': No existe el fichero o el directorio
..DONE..