xtr4nge/FruityWifi #255: Installation bundler error

blankx, 2019-02-07:
I'm running Jessie on a pi3b board using a Alfa awus036nha and I'm continuosly running in to the error "can't install bundler as root" but in order to run the module install script I need to Sudo.
Is there something I'm missing here?

xtr4nge, 2019-07-10:
I always recommend to install it as root and not with sudo, as I had some troubles in the past doing it like that.

Regards,

xtr4nge, 2019-08-02:
Hi blankx,
I hope you are well.
The installer has been updated to support systems based on Debian 10 Buster.

Regards

closed 2019-08-02 by xtr4nge

DICYRUL3S, 2019-12-17:
I'm really sorry, I am new,  I'm using a pi 3b+ i have tried the installation process on three different rasbians. None have installed and gave lots of errors in terminal, mostly permissions denied. Can anyone give me the line they used to install?  I have tried quite a few different commands. None in terminal have worked. The only way I have been able to execute the script is by double clicking on it and running in terminal but it seems to sudo cmds in the script aren't giving it access to pull off the install. I have searched online quite a bit and haven't found anyone else with the same problems