xtr4nge/FruityWifi #257: Adding an extra serivce to kill and restart

hash3liZer, 2019-03-27:
Hi, I am trying to start the access point but the `dhcpcd` service is messing up everything. What i want to do is kill the `dhcpcd` service when the access point is started and then restart it when the access point is stopped. Can you help me or point to the configuration files where i can add these two lines?

hash3liZer, 2019-03-27:
Yes, i finally solved the issue by adding the following two lines in the four starting modes:
```
$ killRegex( "dhcpcd" );
$ exec_fruitywifi( "service dhcpcd start" );
```
in the php configuration file of `ap` module under `www/modules/ap/includes/module_action.php`

hash3liZer, 2019-03-27:
Can you guide me on how to create a new module for this?

xtr4nge, 2019-07-10:
Hi hash3liZer, I hope you are well.
Sorry for the delay in my response. Unfortunately, I never did a proper tutorial for creating modules. 

For now, I would suggest to copy a simple module like Nginx, so you can use it as a baseline:
https://github.com/xtr4nge/module_nginx

Just start changing the folder name to something like "module_dhcpd_killer"
and change in "_info_.php" the variables "mod_name" and "mod_alias", to something like DHCPDKiller, or whatever you want.

Let me know if you have any question, and I will try to help with that.
Regards
xtr4nge

hash3liZer, 2019-07-19:
Got it. Thanks!

closed 2019-07-19 by hash3liZer