xtr4nge/FruityWifi #260: Cannot compile bdfproxy on clean buster install

seanmobrien, 2019-09-01:
I'm trying to install FruityWifi on a Pi3B+ with a fresh install of Buster, but run into issues compiling the mitmproxy sub-module of bdfproxy.  Logs show a couple warnings...

osslsigncode.c:453:36: warning: implicit declaration of function ‘M_ASN1_IA5STRING_new’; did you mean ‘ASN1_IA5STRING_new’? [-Wimplicit-function-declaration]
   info->programName->value.ascii = M_ASN1_IA5STRING_new();
                                    ^~~~~~~~~~~~~~~~~~~~
                                    ASN1_IA5STRING_new

and errors:
osslsigncode.c: In function ‘add_timestamp’:
osslsigncode.c:612:14: error: storage size of ‘mdctx’ isn’t known
   EVP_MD_CTX mdctx;
              ^~~~~
osslsigncode.c:624:3: warning: implicit declaration of function ‘M_ASN1_OCTET_STRING_set’; did you mean ‘ASN1_OCTET_STRING_set’? [-Wimplicit-function-declaration]
   M_ASN1_OCTET_STRING_set(req->messageImprint->digest, mdbuf, EVP_MD_size(md));
   ^~~~~~~~~~~~~~~~~~~~~~~
   ASN1_OCTET_STRING_set
osslsigncode.c: In function ‘cleanup_lib_state’:
osslsigncode.c:819:2: warning: ‘ERR_remove_thread_state’ is deprecated [-Wdeprecated-declarations]
  ERR_remove_thread_state(NULL);
  ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/openssl/e_os2.h:13,
                 from /usr/include/openssl/err.h:13,
                 from osslsigncode.c:106:
/usr/include/openssl/err.h:260:1: note: declared here
 DEPRECATEDIN_1_1_0(void ERR_remove_thread_state(void *))
 ^~~~~~~~~~~~~~~~~~

which at least seem to imply the mitmproxy module is using a pre-1.1 version of openssl, which I can't for the life of me figure out how to get installed onto Buster (and tbh I'm pretty sure I wouldn't want it installed if I could do it :).  Is anyone else running into this or am I missing something in my setup process?  Any aid is greatly appreciated

xtr4nge, 2019-09-21:
Hi seanmobrien,
I will take a look to this. BDFproxy is an old module that needs a lot of dependencies.
I will check as soon as I can.

regards,

H4xl0r, 2020-02-09:
openssl error 
related to tls 1.0 ? ssl package DEPRECATEDIN