xtr4nge/FruityWifi #267: An error occurred with RealTek RT8812AU

kinggx9, 2020-01-28:
Everything is perfect with the Mediatek chipset
Not perfect with RealTek RTL8812AU chipset:
Access point without internet connection
I have this problem at Raspberry Buster, Kali linux ARM and Kali linux x86_64

krapfie, 2020-01-31:
I had the same issue.
Are you using an Alfa AWUS036ACH?

Try to fix that with:

`airmon-ng check kill`

and set up your own DHCPD for the corresponding interface

kinggx9, 2020-01-31:
Using "airmon-ng check kill" is not effective for me.
The setting up DHCPD I don't know
Can you elaborate on how to set up DHCPD
I have a terrible experience with the RTL8812AU chipset (Alfa AWUS036ACH)

caseykey, 2020-02-09:
> I had the same issue.
> Are you using an Alfa AWUS036ACH?
> 
> Try to fix that with:
> 
> `airmon-ng check kill`
> 
> and set up your own DHCPD for the corresponding interface

Do you mean DHCP_Client or DHCP_Server? There are no Debian docs for DHCPD.

kinggx9, 2020-02-16:
I used a new device
It is mt7612u
I am having trouble with the kernel >> 4.19.20
I use it normally with the kernel << 4.19.19
If you have a better idea, please share it with me

midjdjcffnfvf, 2023-06-27:
i have the same problem i make a fake wifi point but there is not internet