xtr4nge/FruityWifi #279: Fruity wifi does not work on Raspberry 4 with Kali 2020.4 (which is based on debian Buster) + alfa awus1900

gopala410, 2020-12-02:
Hi, 
Ive tried to install fruitywifi on the hardware mentioned above. Ill try to be as helpful and precise as I can. 
NOTE: Everything was done with root user
1- downloaded kali 2020.4 from official page and expanded the distro at the sd card for the raspberry 4
2- full updated the kali distro
3- Installed alfa awus1900 card drivers. This is the alfa interface with 4 antennas. 
- Tried to install first downloading the driver and trying to compile from https://github.com/aircrack-ng/rtl8812au and did not worked so when I was tired of f*** around with it the next action worked as a charm. No errors, no issues...
-     Installed with no issues #  apt install realtek-rtl88xxau-dkms
4- Downloaded and installed fruitywifi from https://github.com/xtr4nge/FruityWifi/

- This did not worked as php 7.3 is not the same version as the php included in kali 2020.4 (php 7.4.11). So I could not see the web interface of fruitywifi (the error was something like 505 page not found timeout whatever from nginx version whatever thing).

- So I was smashing my head agains the wall for ages until I got an idea while inspecting the installation script. I changed the version 7.3 by 7.4 in that script and at the end of the script I also changed the way that the script starts nginx and php taking into account that in debian buster this does not work as in the installation script
Im attaching my modified install script. Just remove the .zip (I had to rename it to be able to attach it. So it is not zipped)
[install-FruityWiFi.sh.zip](https://github.com/xtr4nge/FruityWifi/files/5629997/install-FruityWiFi.sh.zip)
Basically the changes are: 
Begining of the script:
**#fruitywifi_php_version="PHP7"
php_version="7.4"**

End of the script:
**# RESTART NGINX + PHP7-FPM
#/etc/init.d/nginx restart
#/etc/init.d/php$php_version-fpm restart
systemctl start nginx
systemctl start php7.4-fpm.service**

5- With that I was able to see the fruitywifi web interface and login as admin
6- Installed the install-modules.pỳ
7- From now on I was able to see all modules not installed so I installed them from the web interface
8- Now all modules are disabled and there is no way to enable them
9.- I cant see my interfaces at the "config"  label in fruitywifi web interface
10- I see my realtek interface from kali OS and I can even connect to my own ssid. See the led blinking, can navigate, even quite slow (strange the broadcom works better than the alfa awus1900 which works fine on windows, so no probs with the interface)

NOTE: I setup US country on fruitywifi even Im not in the US today. This should not be an issue "I hope" for fruity wifi and the interface
NOTE: I think that the fruitywifi installation complained about python-scapy even I have python3-scapy installed on my kali, actually it comes built-in on kali the newer version.

Help will be appreciated.
Please try to be explicit and accurate/concise as I was on my explanation, because Im not a developer and Im starting with this so Im not an expert.

Jmolenberghs, 2020-12-13:
I have the same issue.
Any update on this?

PeaceTrain, 2020-12-15:
great thx this way it works

gopala410, 2021-01-05:
I changed to pumpkin wifi and airgeddon
fruity definitely does not work for up to date kali distributions

zidk, 2021-01-08:
I'm In the same situation, pi3b+ awus036ach Kali 2020.4

gopala410, 2021-01-10:
Looks like this is not maintained anymore. 
Guys try pumpkin wifi and airgeddon., These ones work